maxosprojects/open-dobot #2: Feature request: support for endstop switch/photo interrupter

maxosprojects, 2016-03-27:
There are no sensors to enable completely automatic arm initialization.
As a simplest solution to that problem support for endstop switch/photo interrupter is proposed.

Here is an outline of a proposed solution:
1. Add a command to start base rotation at specified speed that must automatically stop the arm upon a pin state change
1.1. Desired end state change specified in the command
1.2. Desired pin to be polled specified in the command out of predefined unused set of pins available on Arduino
2. Add a command to poll for desired pin state to find when the the traverse finishes
3. Add a command to cancel any traverse commands currently being executed

NikolasE, 2016-03-27:
I'd implement it slightly different:
The user specifies two speeds (fast and slow). (If the endstop is not already triggered,) the robot moves with the fast velocity until the endstop is triggered. It then changes the direction of the movement and moves with the slow speed until the endstop is released. This should speed up the homing sequence while producing a good accuracy even if the pin has to be polled.

maxosprojects, 2016-03-27:
Agree

maxosprojects, 2016-04-01:
Pushing this to 0.5.0 to release SDK initialization from accelerometers. Initializing arm from specific pose wasn't a feasible requirement to comply with and was a huge impediment to do ANYTHING.

Will add endstop support first thing for 0.5.0

commit 6837f712 (2016-04-02): This closes #2. Added limit switch/photointerrupter support with joint calibration routine. Added example how to use it. Added EmegrencyStop() function
closed 2016-04-02 by maxosprojects