maxosprojects/open-dobot #4: Example code for moving to position

NikolasE, 2016-03-28:
so far, there is only the demo to move the arm in a circle but no demo how to move it to a specific position.

maxosprojects, 2016-03-29:
Will take Mike's IK https://github.com/mikef522/open-dobot-gui and create a simple example.
To have an example sooner I will have to simplify it:
1. it will not use accelerometers and would assume predefined pose
2. it will not move in a straight line, but only to get end effector from point A to point B driving each motor as fast as possible.

In order to make the example complete it would be required to:
1. write up accelerometers calibration instructions and implement some helper code to make calibration procedure simple 
2. in code, slice the trajectory into segments, taking into account acceleration/deceleration, to make the end effector move in a straight line from point A to point B

As a bonus, provide an example of arm doing some fancy trajectory, like drawing something in the air.

Needless to say that proper example would take a bit more time. So, I'm planning to make a simple one first for 0.3.0 and then proper one for 0.4.0, unless someone helps.

Currently I'm fixing the protocol to enable command buffer and do crc properly.

mikef522, 2016-03-29:
Hey Nikolas, like Max said, I wrote up a python gui app that does just that (see the repository Max linked to). I uploaded some videos that give a quick demo and explanation here:

Demo videos here:
Set up and move to point demo: https://youtu.be/5oExIlkn5E
Inverse Kinematics Quick Explanation: https://youtu.be/Nsgcq8Uz_Vc
Inverse Kinematics C++ 3D simulation Demo: https://youtu.be/5oExIlkn5EA

I am using ramps though. I look forward to seeing Max's improvements!