maxosprojects/open-dobot #8: Firmware for Baudrate 9600

dborgesr, 2016-03-31:
Was wondering if it would be possible to throw up a precompiled firmware w/ baudrate 9600 for us late adopters?

maxosprojects, 2016-03-31:
The firmware is 115200, which is one of the standard ones (available in any application/driver) and transmits data faster than 9600.
Is there a reason to do 9600?

dborgesr, 2016-03-31:
I got totally confused (i'm a noob / the documentation they provide, well is barely english), they were changing the rate to the bluetooth.
I'm a complete arduino noob as well, way in over my head but I figured this project is worth learning.
Just got it to move! ITS ALIVE

maxosprojects, 2016-03-31:
So I guess it can be closed now. Please reopen if the problem is still not resolved.

closed 2016-03-31 by maxosprojects