maxosprojects/open-dobot #10: moveWithSpeed behavior

schmurfy, 2016-04-17:
Thanks for sharing this great project freeing the dobot from the stupid decisions of its creators, I managed to make the arm move after flashing your firmware into the arduino but I am not reall sure how things are supposed to work.

My initial understanding was that the accelerometers were used to get an approximation of the arm position which seems confirmed by what is shown when connecting to the dobot (I am using `example-sdk.py` as base with most code commented) but as soon as I try to make it move I am lost.
moveWithSpeed seems to take absolute coordinates but if I execute two times the same line the arm moves two times when I expected to only move once and stay in position.

I got even more lost when I try to think of it as relative positioning, it does not either for me, so... I have no clue how it is supposed to work :(

PS: I also tried moveTo but it does not run, I get an undefined method.

maxosprojects, 2016-04-17:
Correct, the accelerometers are used to get approximate pose upon firmware initialization. initialization occurs every time Arduino is reset. In case of linux and mac (I have not tested it in windows) Arduino reset every time the serial port is opened (every time to execute an example).

Also correct about absolute coordinates. They are not relative. Well, the base angle is assumed 0 upon initialization as there are no sensors on it. The reference frame can be seen on https://github.com/maxosprojects/open-dobot/blob/master/docs/images/reference-frame.png
So, given the base angle is zero at start the Y coordinate is also assumed zero at start. So the only coordinates to be read from accelerometers are X and Z.

And your another correct assumption, if I got it right, is that the same command repeated two times should move the arm only once. This is because when the first command finishes, the next command should move the arm to the tool to the same location, which has been already achieved  by the first command. Not sure what you mean by line there, I hope I got it right.

Apologies to moveTo. This is kind of leftover. May be used later when command stitching is implemented (when you could issue multiple commands that would be stitched together into one smooth trajectory without intermediate stops).

There, of course, might be a bug there that simply prevents your dobot from reaching the coordinates you asked it to. Could you share the code you are using to move the arm that isn't working as expected? Don't forget to use proper code block formatting.

schmurfy, 2016-04-18:
Thanks for the detailed answer, no need to apologies I was just trying to understand how this works and wondered what the difference was between `moveTo` and `moveWithSpeed` :)
I realized later that, judging by the image showing how the axis system was mapped that the rotation was not reported by any sensor but the behavior I got is still odd.

Here is the code I used:

```python
from DobotSDK import Dobot
import time

# Maximum speed in mm/s
speed = 700
# Acceleration in mm/s^2
acceleration = 100

dobot = Dobot('/dev/tty.usbmodemFD131', debug=True)
dobot.moveWithSpeed(190.0, 0.0, 50, speed, acceleration)
```

What I couldn't figure out, and still can't after your answer is why the arm move each time I run this code without unplugging the usb cable, I tried the calibration script and what is reported seems fine, I get an approximately correct angle for each arms.
I might have done something wrong but can't really see what it could be.

I will do some more test later today maybe I overlooked something, I have other questions but I will ask them on your forum since they are clearly not an issue ^^

PS: I am using the latest (0.6) firmware on the arduino mega.

schmurfy, 2016-04-18:
I just confirmed that the code I pasted above move the arm everytime I run it, I didn't touch the USB connection or the power on the arm controller.
Any idea what might be wrong and what I could look for ?

schmurfy, 2016-04-18:
nevermind, got it...
I plugged the motor in the wrong position which explain the chaotic behavior, when plugged correctly it works much better :)

closed 2016-04-18 by schmurfy

maxosprojects, 2016-04-18:
fyi, according to what I said above the initialization happens every time you run a script because every time the serial port is opened, which triggers Arduino's bootloader to reset the controller. While executing the same command twice in the same script (actually repeating the same line twice IN the script) should not move the arm, executing the script twice may move the arm every time. This happens becuase the accelerometers are not very accuarate and the estimated tool pose may differ between executions as the accelerometers are re-read on every controller reset (as part of the setup() ). This means that even with the motors hooked up correctly you may see a slight move, don't get confused.

Good that you figured the motors out.

schmurfy, 2016-04-18:
I saw the slight move but overall the positions seems now stable, I forgot that the arduino was reset everytime time the serial port was open though, thanks for pointing that out.