maxosprojects/open-dobot #11: feature request: bluetooth support

schmurfy, 2016-04-18:
it would be nice to be able to remote control the arm :)

maxosprojects, 2016-04-18:
Planned for 0.7.0, unless something more important comes up.

schmurfy, 2016-04-18:
I can help on this if I figure out how their board is layed out, there are a few other things I would like to play with like the air pump grabber, I imagine this is mostly a case of finding how it is wired to the arduino.

maxosprojects, 2016-04-18:
For sucker pump look at dobot.h, it is all documented there. Take laser implementation as a reference. Bluetooth is totally different. It's another serial port that needs to be referenced instead of the USB-serial. Just another register name (actually, another register number).

maxosprojects, 2016-04-19:
It looks like the BT module on dobot is crippled in that it does not support AT commands and there seems no documentation available as to how to configure it using binary command mode.

Originally it was configured to do 115200 baud (well, actually it seems like 119000 more, but 115200 works too). So if you haven't "upgraded" to 9600 version of the proprietary firmware then you could use the original module. But if you did upgrade, then you out of luck, you would have to replace it. Something like http://www.ebay.com/itm/programmable-Bluetooth-Transceiver-compatible-with-Arduino-and-microcontrollers-/182088457543?hash=item2a65515d47:g:5B4AAOSwiylXCs9- would do.

The problem is 9600 baud is too slow to send low-level commands (the ones that FPGA takes) and I haven't found a way to bring it back to 115200.

Currently investigating how to configure proper module that supports AT commands to have instructions and some code ready to do that by the release. I'm thinking of creating a command that would relay AT commands to the BT module to enable to use any module. Maybe this is an overkill.

schmurfy, 2016-04-19:
My development machine is a MacBookPro so I did not even looked at their software which is not available on os x, I should still have a 115200 baud bluetooth module.

maxosprojects, 2016-04-21:
Communication with the BT module has been implemented. However, the module they installed, or the way they configured it, or it is just specifics of BT that makes it not suitable for real-time communication. There is buffering in place that prevents data from coming through right away but rather in bursts of buffered data once in 250ms, which is not suitable for a protocol that acknowledges received data.

This means that controller receives data only 250ms after it has been sent and acknowledges that data. That acknowledgement, in turn, is also delayed by 250ms. So you can send a command only every 500ms (0.5 second), which makes it impossible to have smooth moves given every command covers only 20ms.

I have to defer this feature one release later and investigate alternative BT modules' behavior and options we might have.