maxosprojects/open-dobot #12: controlling the gripper

Elakamons, 2016-04-19:
Even though the gripper isn't that good it would be nice to be able to program it, maybe build a new one but use the same servo pin. Do you know what pin it is that is connected to the servo Grab pin? 

(Not sure if this became as a Pull req)

maxosprojects, 2016-04-20:
The gripper has two servos that are controlled with PWM signals generated by FPGA, so there is no mapping to an Arduino pin. The signals are generated off the four bytes (two bytes per servo) following the "motors direction" byte in the command sent to FPGA. Have a look at dobot.h/sequenceRest structure if you're curious. By simple changing those bytes in that sequenceRest array and reflashing Arduino you can see that working.
The actual implementation is planned for upcoming release.

This cannot become a PR as you haven't pushed any code or created an actual PR.

maxosprojects, 2016-04-28:
See example-gripper.py

commit a4664993 (2016-04-28): This closes #12
closed 2016-04-28 by maxosprojects