maxosprojects/open-dobot #18: OSX support

roykolak, 2016-05-01:
Thanks for this awesome project. Just got a dobot and will be updating it to using your firmware. My q is, do you know if your python sdk will work if it is used on osx? I think I will try to update the firmware on windows because I don't see any other way, but I'm hoping to use this arm primarily on OSX.

maxosprojects, 2016-05-01:
My main development environment is OS X. Currently only Python 2.7 is supported, so as long as you got that everything should work out of the box.

There is an option to flash the firmware in a *nix OS if you have Arduino IDE installed or just avrdude - have a look at the https://github.com/maxosprojects/open-dobot/blob/master/firmware/fpga/upload-external-hex.sh script. All you need to do is to point it to where Arduino/avrdude is installed (the default Arduino location in OS X is already in the script) and the serial port name that Arduino shows up in your system (depending on your laptop's model only the number may change).

roykolak, 2016-05-01:
Oh boy, that made my night. 

I updated the device location in `/dev` in the script, but looks like I supply the path to a .hex. That hex is included in the repo?

maxosprojects, 2016-05-01:
Yep, under releases - https://github.com/maxosprojects/open-dobot/releases - take the latest

roykolak, 2016-05-01:
Awesome! worked like a charm

maxosprojects, 2016-05-01:
Close this issue when you feel it is resolved

closed 2016-05-01 by roykolak