maxosprojects/open-dobot #19: calibration-tool.py instructions

roykolak, 2016-05-01:
"Remove accelerometers from the arm and put them on a flat surface that has no inclination"

Not sure if I follow, the accelerometers seem to be attached firmly to the arm joints and are not removable

NikolasE, 2016-05-01:
They are attached firmly, you have to loosen the screws of the horizontal bars that keep the arm together. Then you can remove the sensors. It's rather delicate, be careful not to damage the PCB's. 
And yes, don't forget to thang the guys from Dobot that had this genius idea how to calibrate their arm.

maxosprojects, 2016-05-01:
Strictly speaking, it is not a hard requirement to calibrate accelerometers but rather a desired procedure to improve accuracy as long as the accelerometers are poorly soldered to their PCBs (have an inclination of two or more degrees). The accelerometers shipped with dobot are not too accurate themselves - according to the datasheet there is about 3-4% error, which, for 90 degrees range, gives about 2.7-3.6 degrees error.

In my case, for the expected value of 1024 at zero inclination I got 1027, which is ```((1027-1024)/500)*90 = 0.54``` degree adjustment, where 500 is the output range (~500-1024 are the values accelerometer outputs for angles 90-0). You can see that the adjustment I got for my accelerometers is five to seven times less than accelerometer's error itself. If you add the looseness of all the joints the adjustment becomes negligibly small, so you can ignore the calibration procedure altogether as it won't make much difference.

maxosprojects, 2016-05-04:
I assume this issue is resolved. Please re-open if needed.

closed 2016-05-04 by maxosprojects