maxosprojects/open-dobot #22: Hardcoded  device name

NikolasE, 2016-05-07:
In most example scripts, the dobot is expected to be at /dev/tty.usbmodem1421. 

At least in Linux, it's straight forward to create a udev-rule so that the dobot (in fact all Arduino Megas)
will be found at /dev/dobot

Just create this file /etc/udev/01-dobot.rules and fill it with

ATTRS{idVendor}=="2341",  ATTRS{idProduct}=="0042",  SYMLINK+="dobot"

Use this to apply the new rule (no need to reboot or unplug the dobot)
sudo udevadm trigger

now, a new device called /dev/dobot should exist:

[~] ls -al /dev | grep do
lrwxrwxrwx   1 root root             7 May  7 16:51 dobot -> ttyACM0

This enables the test scripts to run out of the box without having to check the device name.

maxosprojects, 2016-05-07:
I'm not sure what to do with this issue. I think it belongs to wiki. Could you please create a wiki page and link it in README.md in "Installation" section?

NikolasE, 2016-05-07:
I wrote a wiki page, however I don't have the rights to create a new wiki page

maxosprojects, 2016-05-07:
Fixed. You're now a collaborator and have permissions to edit wiki.
Maybe I should enable editing to anyone at some point, I'm not sure if that is a good idea, I have no experience in wiki area.

closed 2016-05-07 by NikolasE

schmurfy, 2016-05-09:
as you can't do the same as far as I know on OS X what about just extracting the device name in a external and non versioned file ?
This is not a major issue but would solve the fact that each one of us potentially has "permanent" local changes with the right device name.

maxosprojects, 2016-05-10:
Sure. https://docs.python.org/2/library/configparser.html should do. Go ahead and create an improvement ticket.

schmurfy, 2016-05-10:
ok I will have a look at it.