maxosprojects/open-dobot #25: don't reset arduino on disconnect

schmurfy, 2016-05-24:
I was just thinking about this and following my thoughts realized that it is possible to connect using the serial port to an external system without trigerring a reset (DTR), in fact I already did that at work.
I never did it with an arduino but since it is tied to the DTR signal I don't see why it would not work.

I tried to find the corresponding code for pyserial and it seems to be that:

```python
s = serial.Serial(...)
s.setDTR(level=False)

# someone also mention this, not sure if both are needed:
s.dsrdtr=False
```
[source on stackoverflow](http://stackoverflow.com/questions/11385915/serial-interface-initialisation-with-arduino-and-c)

I can't test it at the moment and I am not sure when I will be able to but If post it here in case someone can test before me.

maxosprojects, 2016-05-24:
On Mac I have never seen Arduino reset on disconnect.

Tested this on Mac - has no effect. Also tried ```s.dsrdtr=False```, ```s.dsrdtr=True```, ```s.dtr=False``` (as setDTR has been deprecated since pyserial 3).

There is one caution to not resetting Arduino - with FPGA board the accelerometers are read only on boot (after Arduino resets) because at that time Arduino is SPI master. After accelerometers are read Arduino becomes SPI slave as FPGA rules the bus (clock and stuff). I haven't found a way to make 2560 to become SPI master again and according to google there isn't one. This means that if you do succeed in disabling reset you won't be able to re-initialize the pose from accelerometers and, given there never any slips happen, would have to read the state from Arduino (open-dobot monitors step counts in firmware and you can read them from there with ```GetCounters()``` in DobotDriver).

schmurfy, 2016-05-25:
I couldn't remember if this was when connecting or disconnecting so I added both xD

if I understood how it work correctly reading the accelerometers is only used to deduce the actual position on start, right ? Because if that is indeed the case I don't understand your warning, one thing I would like, and preventing the reset would allow it, is for the arduino firmware to keep track of position between resets, especially for the rotation since there is no sensor to tell the current position.

Are any of my assumptions wrong ?

maxosprojects, 2016-05-25:
Accelerometers are read when firmware starts (boots, after Arduino resets), not when your python code starts. Currently SDK asks firmware "what were the values you read from accelerometers when you booted up?". It takes those values, converts them to angles and uses angles to estimate the pose in steps (starting position). It also sets those steps in firmware that also keeps track of current position.
When you start your python code it connects to Arduino which usually resets Arduino and you get current pose. If Arduino is not reset then you will read outdated accelerometer values. This can be addressed by reading steps instead of accelerometers in SDK, but only in case you know for sure that Arduino has not been reset (if it is at all possible on all platforms).

maxosprojects, 2016-06-01:
Closing this issue as there is no further discussion.

Please re-open if you have any updates to this topic.

closed 2016-06-01 by maxosprojects