maxosprojects/open-dobot #26: Firmware change to support MP65 accelerometer

rgrbic, 2016-05-25:
The dobot arm from v1.2 comes with MP65 accelerometers. The current firmware version (0.5.0) does not properly read these accelerometers.

maxosprojects, 2016-05-25:
Current open-dobot version is 1.1.0 https://github.com/maxosprojects/open-dobot/releases/tag/1.1.0

maxosprojects, 2016-05-26:
There are three options here:

1. Someone who has a logic analyzer (BeagleBone can be used as one) or an oscilloscope implements that.
2. Get 2 of these http://www.amazon.com/SMAKN%C2%AE-Arduino-MPU-6050-gyroscope-accelerometer/dp/B00NH8Z6BU/ or similar, solder 5 wires (or 4 wires and 2 resistors) and mount those accelerometers on top of existing ones (hot glue would do).
3. Switch to RAMPS v1.4 and do the same as in option 2.

Looking at the amount of contributions to this project option 1 is not likely to happen any time soon.
Support for MPU-6050 in FPGA version for option 2 can be added within two weeks once someone expresses interest in that option.
Support for MPU-6050 in RAMPS version for option 3 is coming with open-dobot v1.2 (in the next couple of days).

MP65 is most probably a clone of MPU-6050 with "who knows what implementation" and absolutely no documentation (thanks dobot team for using unknown crap). To start with, it looks like it sits on SPI bus, while MPU6050 sits on I2C - in documentation for MPU-60x0 only MPU6000 sits on SPI.
Given I don't have MP65-based accelerometer, guessing the protocol is not likely to get us anywhere.

So I highly recommend switching to MPU6050 - it is a well-known and widely used (e.g. in drones) MPU.

rgrbic, 2016-05-26:
I agree.

maxosprojects, 2017-01-07:
I'm going to look at that https://www.amazon.com/UCTRONICS-MPU-6500-6-Axis-Gyroscope-Accelerometer/dp/B01DIPU152/ . If it appears to be the same chip (MPU-6500) installed on version 1.2 then I will add it to the firmware.

rgrbic, 2017-01-14:
Hi here is picture of mine accelerometer if it helps
[image]

maxosprojects, 2017-01-15:
The label on the chip in the middle is not clear on your picture, but it's ok, other users have uploaded pictures before and it was clear that it is MP65.

afifswaidan, 2020-03-05:
Hello, I am continuing the development on the dobot arm as I am a mechatronics engineering student in my senior year currently. 
I am trying to add realtime accelerometers reporting to use them as a negative feedback loop during the movement of the arm. I edited the firmware to send to me static values that I set for testing whenever I call the getaccelrometers() function from the dobot SDK.
I also added many other features (gui and interactive user input through terminal) and fixed a few math errors in the dobot kinematics but I am new to github and idk how to send the files to you to check them and give me ur feedback.
Thanks!

afifswaidan, 2020-03-05:
I forked ur repository but idk how to share it with you to collaborate. Thanks for your great work by the way! <3