maxosprojects/open-dobot #31: RAMPS wiring

louise46, 2016-07-12:
Hi,
I was wondering how you have connected each of the GY-521 SDA and SCL pins to the same pin 20 and 21 on the RAMPS board using jumper wires - are you using a breadboard? 
Also, are you using a separate power supply or using the +12V from pin D8 and a ground pin on the RAMPS board?

Thanks,
Louise

maxosprojects, 2016-07-12:
You could use a breadboard, but, as the arm is expected to move around a lot, using breadboard is not recommended. I used ribbon cable https://www.sparkfun.com/products/10647 to wire modules together (just soldered), then cut couple of jumper cables and then soldered some more ribbon cable to it and to the module on rear arm.

I'm not sure what you mean about 12V. Separate power supply for what?

Just in case, you are not talking about feeding 12V to the GY-521 modules, do you? Their supply voltage is 5V.

louise46, 2016-07-12:
I meant 12V to power the RAMPS board - you say to connect a 12V supply to the RAMPS MSTBA4 connector. Is this using a separate PSU as the dobot was originally powered through the FPGA?

maxosprojects, 2016-07-12:
Still not sure what you mean by "separate" and what you were going to feed from D8.

In case of driving Dobot with RAMPS you need only one power supply, connected to MSTBA4, which has GND and two lines for 5A and 15A respectively. Both those lines should be connected to the power supply that came with Dobot. Just make sure polarity is right.

closed 2016-07-19 by louise46