maxosprojects/open-dobot #36: self._port.send() Error

cjbayens, 2016-11-04:
Hi, I'm working on editing DobotDriver.py to work with the new Dobot Magician without the need for RAMPS. My question is in the definition of _read, there is a command called self._port.send() but the serial module doesn't have a "send" command. What is this line supposed to do? Thanks!

maxosprojects, 2016-11-06:
It is ```serial_aggregator``` class defined in DobotDriver.py

But before proceeding with your approach take a moment to answer a question: do you plan to flash open-dobot firmware with magician?
I have no idea what hardware magician is built with and I am certain open-dobot firmware won't work there.
open-dobot's application software (e.g. python modules, examples) expects to talk to an Arduino Mega with flashed open-dobot firmware.

maxosprojects, 2017-01-15:
No activity.

closed 2017-01-15 by maxosprojects